gazebosim/garden-tutorial-party #14: docs: troubleshooting

audrow, 2022-08-29:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Source

## Links
- [Live](https://gazebosim.org/docs/garden/troubleshooting)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/troubleshooting.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

srmainwaring, 2022-09-06:
Checked macOS sections on macOS Monterey, MacPro (Intel).

1. No issue with urdfdom if latest brew versions are installed.
2. Using the brew ruby version works. The default shell on macOS is now zsh, so the PATH should be updated in `.zprofile` or `.zshrc`.
3. Add a note that running `brew update && brew upgrade` will almost surely require a clean and full recompile when building from source. This is because brew runs a rolling release system and if cmake or python or the compiler update then there is a good change that all the cmake caches will be invalidated.
4. Extra: it's possible to run gz and ros2 on macOS without disabling SIP. This should be the recommended approach as we should not encourage users to disable security settings. This does require a change in some build settings (namely building with the `-DCMAKE_MACOSX_RPATH=FALSE` flag).

j-rivero, 2022-09-09:
Hey Rhys!

Thanks so much for the testing.

> * 
> 
> * Add a note that running `brew update && brew upgrade` will almost surely require a clean and full recompile when building from source. This is because brew runs a rolling release system and if cmake or python or the compiler update then there is a good change that all the cmake caches will be invalidated.
> 
> * Extra: it's possible to run gz and ros2 on macOS without disabling SIP. This should be the recommended approach as we should not encourage users to disable security settings. This does require a change in some build settings (namely building with the `-DCMAKE_MACOSX_RPATH=FALSE` flag).

Would you mind open a PR for this great suggestions so we don't lose them?