gazebosim/garden-tutorial-party #196: gz-math: C++ Get Started

audrow, 2022-08-29:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/api/math/7/cppgetstarted.html)
- [Source](https://github.com/gazebosim/gz-math/tree/gz-math7/tutorials/cppgetstarted.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tejalbarnwal, 2022-09-13:
Hey, could you please assign this to me?

tejalbarnwal, 2022-09-16:
The tutorial works fine. 
For the live link, I found some separate texts written at the end describing some classes and functions used above in the code. An issue related to a similar matter has already been reported [here](https://github.com/gazebosim/docs/issues/296). I have added a comment to it reporting for this tutorial live link.

audrow, 2022-09-16:
Thanks @tejalbarnwal!

closed 2022-09-16 by audrow