gazebosim/garden-tutorial-party #211: gz-math: Installation

audrow, 2022-08-29:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/math/7/install.html)
- [Source](https://github.com/gazebosim/gz-math/tree/gz-math7/tutorials/installation.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ahcorde, 2022-09-05:
PR to fix install instructions: https://github.com/gazebosim/gz-math/pull/493

srmainwaring, 2022-09-07:
System: macOS Monterey, MacPro (Intel)

Building from source, macOS 

In step 2 *Install dependencies*

The formula is missing a dependency on swig, so the ruby bindings may not build if swig is not already installed.

```zsh
% brew deps --tree gz-math7
osrf/simulation/gz-math7
├── eigen
├── gz-cmake3
│   ├── cmake
│   └── pkg-config
├── gz-utils2
│   └── gz-cmake3
│       ├── cmake
│       └── pkg-config
└── ruby
    ├── libyaml
    ├── openssl@1.1
    │   └── ca-certificates
    └── readline
```