gazebosim/garden-tutorial-party #218: gz-math: Python Get Started

audrow, 2022-08-29:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/api/math/7/pythongetstarted.html)
- [Source](https://github.com/gazebosim/gz-math/tree/gz-math7/tutorials/pythongetstarted.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

scpeters, 2022-09-06:
The tutorial implies that `PYTHONPATH` is only needed if you build from source, but it's required for me when using the homebrew bottle binaries. I think I have a fix in https://github.com/osrf/homebrew-simulation/pull/2036

scpeters, 2022-09-14:
the `PYTHONPATH` issue is resolved, but I have made a small update to the tutorial in https://github.com/gazebosim/gz-math/pull/502

closed 2022-09-14 by scpeters