gazebosim/garden-tutorial-party #300: gz-physics: Introduction

audrow, 2022-08-29:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/api/physics/6/introduction.html)
- [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics6/tutorials/01_intro.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JoanAguilar, 2022-09-06:
I'm assuming that this issue, #301, #302, #303, #304, #305, #306, and #307, are all dups of each other. Nothing really is system or binary/source specific here.

The two things I saw that stood out are:

 - The image in the document refers to `ign` rather than `gz`.

 - There's a "Development timeline" section that I don't know if it's up to date.

audrow, 2022-09-06:
> I'm assuming that this issue, #301, #302, #303, #304, #305, #306, and #307, are all dups of each other. Nothing really is system or binary/source specific here.

Good call, @JoanAguilar. I've closed the other issues.

JoanAguilar, 2022-09-06:
Thanks @audrow!

I'm trying to figure out who might have a non-rasterized version of the [image](https://github.com/gazebosim/gz-physics/blob/gz-physics6/tutorials/img/gz-libraries.png) and who might know if the timeline is accurate. Seeing that this is `gz-physics` maybe @scpeters ?

scpeters, 2022-09-07:
I remember seeing google presentations with these types of graphics, let me see what I can find...

scpeters, 2022-09-07:
that image was added in https://github.com/gazebosim/gz-physics/commit/da61869e1b5d45719d696ce4b623d6d92b5be311, but I can't figure out where it came from. It might be easier to create a new SVG