gazebosim/garden-tutorial-party #41: Start here - Garden Testing Meta Ticket

audrow, 2022-08-29:
> If anything is confusing, please let me know in a comment on this issue

How to use this repo:

1. Find an open issue from the lists below
2. Assign yourself to it (or comment on the issue if you don't have access).
3. Test the live link, if there is one. If there is a link, but it doesn't bring you to where you expect, add the `Live page link doesn't work as expected` label to the issue.
4. If there is a live link, follow along with the documentation or run the code - on the same operating system and with the same build type.
5. If there is a problem, add the `Problem with the testcase` label to the issue and create an issue in the offending repository and make sure to link (mention) the issue you've opened to the issue in this repository. 
   - Feel free to open a PR fixing the issue. 
   - Note that the source link may be helpful in finding where the problem is. If the source link is wrong, add a `Source page link doesn't work as expected` label to the issue.
6. If everything seems correct, close the issue in this repository (or comment on the issue, if you don't have access).
7. Once all of the issues have been closed in one subgroup in the list below, check off that group.

---

Here are some things to know about Github search:
- You can filter available results with a label by typing `label:my-label`. And it works for multiple labels: `label:my-label-1 label:my-label-2`.
- If your label has a space in it, you can put the label in quotes: `label:"my label"`.
- To show results that **don't** have a label, use a minus sign: `-label:no-results-with-this-label`.

Happy testing!

---

- [ ] [Ubuntu Focal, Binary](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22)

  <details><summary>Labels</summary>

  - [ ] [docs](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22docs%22)
  - [ ] [garden](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22garden%22)
  - [ ] [gz_ros2_control](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz_ros2_control%22)
  - [ ] [rolling](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22rolling%22)
  - [ ] [gz-common](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-common%22)
  - [ ] [gz-fuel-tools](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-fuel-tools%22)
  - [ ] [gz-gui](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-gui%22)
  - [ ] [gz-launch](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-launch%22)
  - [ ] [gz-math](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-math%22)
  - [ ] [gz-msgs](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-msgs%22)
  - [ ] [gz-physics](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-physics%22)
  - [ ] [gz-plugin](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-plugin%22)
  - [ ] [gz-rendering](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-rendering%22)
  - [ ] [gz-sensors](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-sensors%22)
  - [ ] [gz-sim](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-sim%22)
  - [ ] [sdf worlds](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22sdf%20worlds%22)
  - [ ] [gz-tools](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-tools%22)
  - [ ] [gz-transport](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-transport%22)
  - [ ] [gz-utils](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-utils%22)
  - [ ] [sdf_tutorials](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22sdf_tutorials%22)
  - [ ] [ros_gz](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22ros_gz%22)
  - [ ] [noetic](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22noetic%22)
  - [ ] [sdformat](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Focal%22+label:%22Binary%22+label:%22generation-1%22+label:%22sdformat%22)

  </details>
- [ ] [Ubuntu Jammy, Binary](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22)

  <details><summary>Labels</summary>

  - [ ] [docs](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22docs%22)
  - [ ] [garden](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22garden%22)
  - [ ] [gz_ros2_control](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz_ros2_control%22)
  - [ ] [rolling](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22rolling%22)
  - [ ] [gz-common](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-common%22)
  - [ ] [gz-fuel-tools](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-fuel-tools%22)
  - [ ] [gz-gui](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-gui%22)
  - [ ] [gz-launch](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-launch%22)
  - [ ] [gz-math](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-math%22)
  - [ ] [gz-msgs](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Ubuntu%20Jammy%22+label:%22Binary%22+label:%22generation-1%22+label:%22gz-msgs%22)
[... cut ...]
  - [ ] [gz-utils](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Windows%2010%22+label:%22Source%22+label:%22generation-1%22+label:%22gz-utils%22)
  - [ ] [sdf_tutorials](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Windows%2010%22+label:%22Source%22+label:%22generation-1%22+label:%22sdf_tutorials%22)
  - [ ] [install](https://github.com/gazebosim/garden-tutorial-party/issues?q=is%3Aissue+is%3Aopen+label:%22Windows%2010%22+label:%22Source%22+label:%22generation-1%22+label:%22install%22)

  </details>

gaileyleseman, 2022-09-12:
Hi Audrow! I have a general question. When trying out the tutorials I encountered a couple issues which are not really related to the tutorials. For example [here](https://github.com/gazebosim/garden-tutorial-party/issues/1985) and [here](https://github.com/gazebosim/garden-tutorial-party/issues/1965). I have now added comments to the tutorials where I encountered these issues, but this is probably not the best way to log this. How should I go about logging these issues? (and/or test if it is something just on my machine or in general)

azeey, 2022-09-12:
Hi @gaileyleseman, the best for non-tutorial related issues is to either create an issue on the corresponding gz library (use gz-sim if you don't know for sure) or to update an existing issue if there is one. Once that's done, it would be great if you can post a comment in the tutorial issue referring to the issue you've created/updated.

peci1, 2022-09-18:
Hi, I can't add labels to issues. What should I do instead of that?

tejalbarnwal, 2022-09-18:
Hey, there are some doubts that I wanted to ask. It may be trivial, but it isn't allowing me to try out some of the tutorials from the documentation. Since I wasn't sure if it was a bug, I asked a question on gazebo answers [here](https://answers.gazebosim.org/question/28448/not-able-to-run-spawn_urdf-tutorial-with-garden/). Could you please look at it whenever you are available?

azeey, 2022-09-19:
> Hi, I can't add labels to issues. What should I do instead of that?

Can you post a comment with what the label should be?

peci1, 2022-09-19:
> Can you post a comment with what the label should be?

Sure, this was a general question. I'll post as needed.

peci1, 2022-09-19:
I see only a single issue for sdformat here. Is that right?