gazebosim/garden-tutorial-party #460: gz-rendering: Installation

audrow, 2022-08-29:
## Setup
- Os: Windows 10
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/api/rendering/7/installation.html)
- [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering7/tutorials/02_install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

peci1, 2022-09-18:
No binary is published on conda forge. Or I can't find it.

peci1, 2024-09-16:
This should be closed. Prerelease binaries are not (yet) distributed on Conda.

azeey, 2024-09-16:
@peci1, this is the garden tutorial party. Did you mean to comment on this? I'll actually go ahead archive this repo since we don't use it anymore and Garden is going to be EOL in a couple of weeks.