gazebosim/garden-tutorial-party #53: gz-common: Installation

audrow, 2022-08-29:
## Setup
- Os: Windows 10
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/common/5/install.html)
- [Source](https://github.com/gazebosim/gz-common/tree/gz-common5/tutorials/install.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

peci1, 2022-09-18:
geospatial doesn't build:

```
CMake Warning at D:/programovani/ign5-ws/install/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:69 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [geospatial]: Missing dependency [GDAL].
      ^~~~~ Set SKIP_geospatial=true in cmake to suppress this warning.

Call Stack (most recent call first):
  CMakeLists.txt:140 (gz_configure_build)

```

peci1, 2022-09-18:
https://github.com/gazebosim/gz-common/pull/438

peci1, 2022-09-19:
Pending new release of docs, this should be working now with https://github.com/gazebosim/gz-common/pull/438 merged.