gazebosim/garden-tutorial-party #562: gz-sensors: Segmentation Camera in Gazebo

audrow, 2022-08-29:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/api/sensors/7/segmentationcamera_igngazebo.html)
- [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors7/tutorials/segmentation_camera.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tejalbarnwal, 2022-09-13:
Hey, could you please assign this to me?

iche033, 2022-09-16:
https://github.com/gazebosim/gz-sensors/pull/269

tejalbarnwal, 2022-09-17:
Hey, I found all of the sensors had a system plugin in the examples [here](https://github.com/gazebosim/gz-sim/tree/gz-sim7/examples/worlds).
The system plugin is given by the following which is a necessity for sensors to work. It would be nice if this could be added to the [Introduction to sensors](https://gazebosim.org/api/sensors/7/introduction.html) or [segmenation camera tutorial](https://gazebosim.org/api/sensors/7/segmentationcamera_igngazebo.html).
```
    <plugin
      filename="gz-sim-sensors-system"
      name="gz::sim::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>
```
Should I make a PR for the same?

tejalbarnwal, 2022-09-17:
Also, assigning the label of the model by adding the plugin to the model's <visual> tag, gives me the following warning 
```
Warning [Utils.cc:129] [/sdf/plugin[@name="gz::sim::systems::Label"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
```

closed 2022-09-19 by audrow

iche033, 2022-09-20:
> The system plugin is given by the following which is a necessity for sensors to work. It would be nice if this could be added to the Introduction to sensors or segmenation camera tutorial mentioning the sensors that use this.

Sure that helps. I think you can mention this in the segmentation camera tutorial.

tejalbarnwal, 2022-09-21:
> Sure that helps. I think you can mention this in the segmentation camera tutorial.

I will do it