gazebosim/garden-tutorial-party #1002: gz-sim: import_mesh.sdf

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim7/examples/worlds/import_mesh.sdf)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

AndrejOrsula, 2022-09-16:
Works.

---

The SDF world still uses <https://fuel.ignitionrobotics.org> instead of <https://fuel.gazebosim.org>. I don't know if you want to change that.

```xml
<uri>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/Electrical Box</uri>
```

I don't see an open PR that would change it, but maybe I missed it. I also don't want to open a new PR only for this instance as there might be more occurrences in other files.

audrow, 2022-09-19:
Thanks, @AndrejOrsula!

closed 2022-09-19 by audrow