gazebosim/garden-tutorial-party #1122: gz-sim: lights.sdf

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim7/examples/worlds/lights.sdf)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

AndrejOrsula, 2022-09-16:
Seems fine, not much to say. Editing XML elements within the SDF file also works and changes the lights in Gazebo correspondingly.

[image]

audrow, 2022-09-16:
Thanks @AndrejOrsula!

closed 2022-09-16 by audrow