gazebosim/garden-tutorial-party #1451: gz-sim: segmentation_camera.sdf

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim7/examples/worlds/segmentation_camera.sdf)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

AndrejOrsula, 2022-09-16:
- [x] Visualization of image topics via GUI works

[image]

- [x] Storing of images also works (after enabling the relevant XML elements as indicated in the SDF file)

[image]

- [x] Home object is correctly considered as a background and not shown in the output
```xml
<!-- Home, we didn't annotate (label) it, so it will be considered as background -->
<include>
  <name>home</name>
  <pose>-15 0 0 0 0 1.57</pose>
  <uri>
    https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Collapsed House
  </uri>
</include>
```

There are only two (very minor) issues.

1. Few textures are not resolved, but it has no effect on the purpose of the tutorial
```log
[Err] [SystemPaths.cc:519] Could not resolve file [pickup_diffuse.jpg]
[Err] [SystemPaths.cc:519] Could not resolve file [wheels2.jpg]
[Err] [SystemPaths.cc:519] Could not resolve file [suv.png]
[Err] [SystemPaths.cc:519] Could not resolve file [wheels_01.png]
[Err] [SystemPaths.cc:519] Could not resolve file [collapsed_house_diffuse.jpg]
```
2. The selector of image topics becomes active only after pressing the refresh button
[image]

nkoenig, 2022-09-16:
I fixed the texture issues. Thanks for the thorough review.

closed 2022-09-16 by nkoenig