gazebosim/garden-tutorial-party #1542: gz-sim: spherical_coordinates.sdf

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Jammy
- BuildType: Source

## Links
- [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim7/examples/worlds/spherical_coordinates.sdf)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jennuine, 2022-09-06:
@ahcorde are you experiencing that crash? I just tried on focal and it's not happening for me

ahcorde, 2022-09-06:
I tried today in `jammy`, I'm not experimenting the crash but the GUI is frozen

I'm compiling from source with `qt5.15.3`.

jennuine, 2022-09-06:
Hm ok, I have `qt5.12.8` on focal (compiled from source)

Can you try replacing this line in `gz-gui`: https://github.com/gazebosim/gz-gui/blob/9b65b810019e239d931de4a050df931295c88328/src/plugins/navsat_map/NavSatMap.qml#L162

with

```
function onNewMessage(_latitudeDeg, _longitudeDeg) {
```

and let me know if the GUI is still frozen?

ahcorde, 2022-09-07:
yes, it's still frozen, I made this change to remove the deprecation. it might be related with https://github.com/msys2/MINGW-packages/issues/11859