gazebosim/garden-tutorial-party #1814: gz-transport: Services

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Focal
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/transport/12/services.html)
- [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport12/tutorials/05_services.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

caguero, 2022-09-09:
Some tweaks from https://github.com/gazebosim/gz-transport/pull/354 but the tutorial works.

caguero, 2022-09-09:
Also reported an issue in https://github.com/gazebosim/docs/issues/296, because there's some weird text rendered in some tutorials.

closed 2022-09-09 by caguero