gazebosim/garden-tutorial-party #1873: sdf_tutorials: Install

audrow, 2022-08-30:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Source

## Links
- [Live](http://sdformat.org/tutorials?tut=install)
- [Source](https://github.com/gazebosim/sdf_tutorials/blob/master/install/tutorial_12.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

azeey, 2022-09-10:
We probably need a tutorial for version `>=13.0` that renames all references of `ignition` to `gz` .

audrow, 2022-09-12:
> We probably need a tutorial for version `>=13.0` that renames all references of `ignition` to `gz` .

Once it exists, feel free to @ me and I'll generate new issues for it.