gazebosim/garden-tutorial-party #1906: sdf_tutorials: Quickstart

audrow, 2022-08-30:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Live](http://sdformat.org/tutorials?tut=quickstart)
- [Source](https://github.com/gazebosim/sdf_tutorials/blob/master/quickstart/tutorial_2-0.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tejalbarnwal, 2022-09-09:
Hey @audrow , I am an undergrad from India. Can I try the tutorial and review it? I am a gazebo-classic user and haven't used gazebo(ignition based). I would love to try this out. Can I work on this?

audrow, 2022-09-09:
> Hey @audrow , I am an undergrad from India. Can I try the tutorial and review it? I am a gazebo-classic user and haven't used gazebo(ignition based). I would love to try this out. Can I work on this?

Sure, I'll assign you to the issue!

tejalbarnwal, 2022-09-13:
Hey, I followed the live link tutorial for both `sdformat 9 and 10` and this works fine! The only thing that concerns me is that when I try to run `./check_sdf model.sdf` it says that `model.sdf is not a valid SDF file!`, whereas if I run the same SDF file with a different name, it runs fine. Am I missing something here?

audrow, 2022-09-13:
> The only thing that concerns me is that when I try to run `./check_sdf model.sdf` it says that `model.sdf is not a valid SDF file!`, whereas if I rename the same SDF file with a different name ( something other than model), it runs fine. Am I missing something here?

I don't think so. I think maybe it has something to do with how `HasElement` works. Would you open an issue on the https://github.com/gazebosim/gz-sim and then link the issue here?

tejalbarnwal, 2022-09-16:
Hey, I created an issue [here](https://github.com/gazebosim/gz-sim/issues/1715)

audrow, 2022-09-16:
Thank you @tejalbarnwal!

I'm going to close this issue, as I think https://github.com/gazebosim/gz-sim/issues/1715 isn't necessarily related to this tutorial.

closed 2022-09-16 by audrow