gazebosim/garden-tutorial-party #1963: docs: hotkeys

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/hotkeys)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/hotkeys.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tejalbarnwal, 2022-09-12:
Hey, please assign this issue to me. While I was acquainting myself with the basics of garden, I was going through the tutorials listed on its documentation site. I found the following stuff that might be looked into:
1. a very minor change concerns the letter "q" for closing the main window in the description
[image]
2. I faced the following exception when I tried to load a custom-made config file into the sim. I have attached the screen video here.
[image]

Please help me to resolve the same

tejalbarnwal, 2022-09-16:
- Commented on the issue already reported [here](https://github.com/gazebosim/gz-gui/issues/489) for Jammy 
- The source link for the tutorial shows the updated documentation for the close button, but it is not reflected on the live link.