gazebosim/garden-tutorial-party #1977: docs: install_osx

audrow, 2022-09-06:
## Setup
- Os: MacOS 11 (Big Sur)
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/install_osx)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/install_osx.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

azeey, 2022-09-08:
Installation works, but running `gz` results in the following error message:

```
I cannot find any available 'gz' command:
        * Did you install any Gazebo library?
        * Did you set the GZ_CONFIG_PATH environment variable?
            E.g.: export GZ_CONFIG_PATH=$HOME/local/share/gz
```

azeey, 2022-09-08:
https://github.com/osrf/homebrew-simulation/pull/2039

closed 2022-09-09 by azeey