gazebosim/garden-tutorial-party #1978: docs: install_ubuntu

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Focal
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/install_ubuntu)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/install_ubuntu.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

j-rivero, 2022-09-08:
Works

closed 2022-09-08 by j-rivero

tejalbarnwal, 2022-09-21:
Hey, though the instructions do work but the some of tools are missing from the command line like `gz service` and `gz topic`
When I try to do `gz --help`, the output gives me
```
The 'gz' command provides a command line interface to the Gazebo Tools.

  gz <command> [options]

List of available commands:

  help:          Print this help text.
  gui:           Launch graphical interfaces.
  fuel:          Manage simulation resources.
  sim:           Run and manage the Gazebo Simulator.
  gazebo:        Deprecated. Alias for sim.
  sdf:           Utilities for SDF files.
  model:         Print information about models.
  plugin:        Print information about plugins.
  msg:           Print information about messages.
  log:           Record or playback topics.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'gz help <command>' to print help for a command.

```

While, I also tried installing garden using source on Focal, the same command upon following the installation tutorial give me the following output with `gz --help`
```
The 'gz' command provides a command line interface to the Gazebo Tools.

  gz <command> [options]

List of available commands:

  help:          Print this help text.
  gui:           Launch graphical interfaces.
  fuel:          Manage simulation resources.
  topic:         Print information about topics.
  service:       Print information about services.
  sim:           Run and manage the Gazebo Simulator.
  gazebo:        Deprecated. Alias for sim.
  sdf:           Utilities for SDF files.
  model:         Print information about models.
  launch:        Run and manage executables and plugins.
  plugin:        Print information about plugins.
  msg:           Print information about messages.
  log:           Record or playback topics.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'gz help <command>' to print help for a command.

```
Am not sure why does my binary installation doesn't give command line commands for topic, launch, service, etc. I did follow the instructions given on the live link.