gazebosim/garden-tutorial-party #1979: docs: install_ubuntu

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Jammy
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/install_ubuntu)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/install_ubuntu.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

gaileyleseman, 2022-09-09:
The instructions for installation and uninstalling work as expected :)

When running `gz sim` I do get some warnings/errors, for example when selecting the Panda Joint Control World: 
- `Warning [Utils.cc:129] [/sdf/model[@name="Panda"]/link[@name="panda_leftfinger"]/visual[@name="panda_leftfinger_visual"]/material/pbr/metal/metallic_map:/home/gailey/.gz/fuel/fuel.ignitionrobotics.org/openrobotics/models/panda with ignition position controller model/2/model.sdf:L624]: XML Element[metallic_map], child of element[metal], not defined in SDF. Copying[metallic_map] as children of [metal].`
- `[Err] [SystemPaths.cc:519] Could not resolve file [texture.png]`
- `libEGL warning: DRI2: failed to create dri screen`
  - this is probably related to my machine

audrow, 2022-09-13:
Thanks, @gaileyleseman!

closed 2022-09-13 by audrow