gazebosim/garden-tutorial-party #1980: docs: install_windows

audrow, 2022-09-06:
## Setup
- Os: Windows 10
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/install_windows)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/install_windows.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

scpeters, 2022-09-21:
There is a post on gazebo answers having trouble running installing on windows with Conda:

* https://answers.gazebosim.org/question/28447/issue-with-running-on-windows-with-miniconda3-install/

mayman99, 2022-09-21:
I'm working on windows nowadays for a project, I could give this a run :+1: