gazebosim/garden-tutorial-party #1985: docs: Manipulating_models

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Jammy
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/Manipulating_models)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/Manipulating_models.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

gaileyleseman, 2022-09-10:
Live page link doesn't work as expected
https://gazebosim.org/docs/garden/Manipulating_models -> https://gazebosim.org/docs/garden/manipulating_models

gaileyleseman, 2022-09-10:
'Translate mode' works as expected, but in 'Rotate mode', the circles to manipulate the orientation of the model do now show up? 
[image]

gaileyleseman, 2022-09-10:
`Pose` attribute is empty for all entities. 

[image]

gaileyleseman, 2022-09-10:
Other notes: 
- selecting multiple entities with `Ctrl` seems to only work when clicking in the Entity Tree, not in the scene

audrow, 2022-09-13:
Thanks @gaileyleseman!