gazebosim/garden-tutorial-party #1992: docs: Model_insertion_fuel

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Focal
- BuildType: Source

## Links
- [Live](https://gazebosim.org/docs/garden/Model_insertion_fuel)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/Model_insertion_fuel.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jasmeet0915, 2022-09-11:
The Live link in the issue above seems to be invalid. Here's the correct one: https://gazebosim.org/docs/garden/fuel_insert

audrow, 2022-09-12:
Thanks for pointing this out, @jasmeet0915. I guess this repo breaks the URL pattern - I'll make a note on other issues.

jasmeet0915, 2022-09-12:
@audrow 
Also, I have tested this tutorial and it works perfectly as expected! 

[image]

audrow, 2022-09-13:
Thanks @jasmeet0915!

closed 2022-09-13 by audrow