gazebosim/garden-tutorial-party #2033: docs: spawn_urdf

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Jammy
- BuildType: Binary

## Links
- [Live](https://gazebosim.org/docs/garden/spawn_urdf)
- [Source](https://github.com/gazebosim/docs/tree/master/garden/spawn_urdf.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

gaileyleseman, 2022-09-09:
`gz service -l` is an invalid command, this should probably be `gz sim service -l`? However this is not listed as an option when you run `gz help sim`. Not sure how to proceed with this tutorial without the service command. 

Output for `gz sim service -l`: 
```
/usr/lib/ruby/gz/cmdsim7.rb:323:in `parse': ambiguous option: -l (OptionParser::AmbiguousOption)
	from /usr/lib/ruby/gz/cmdsim7.rb:337:in `execute'
	from /usr/bin/gz:308:in `<main>'
/usr/lib/ruby/gz/cmdsim7.rb:323:in `parse': invalid option: l (OptionParser::InvalidOption)
	from /usr/lib/ruby/gz/cmdsim7.rb:337:in `execute'
	from /usr/bin/gz:308:in `<main>'
```

Output for `gz help sim`:
```
Run and manage Gazebo simulations.                                              
                                                                                
  gz sim [options] [file]                                                       
                                                                                
                                                                                
Available Options:                                                              
  -g                           Run only the GUI.                                

  --iterations [arg]           Number of iterations to execute.                 

  --levels                     Use the level system. The default is false,      
                               which loads all models. It's always true         
                               with --network-role.                             

  --network-role [arg]         Participant role used in a distributed           
                               simulation environment. Role is one of           
                               [primary, secondary]. It implies --levels.       

  --network-secondaries [arg]  Number of secondary participants expected        
                               to join a distributed simulation                 
                               environment. (Primary only).                     

  --record                     Use logging system to record states and          
                               console messages to the default location,        
                               in ~/.gz/sim/log.                       

  --record-path [arg]          Implicitly invokes --record, and specifies       
                               custom path to put recorded files. Argument      
                               is path to record states and console             
                               messages. Specifying this argument will          
                               enable console logging to a console.log          
                               file in the specified path.                      

  --record-resources           Implicitly invokes --record, and records         
                               meshes and material files, in addition to        
                               states and console messages.                     

  --record-topic [arg]         Specify the name of an additional topic to       
                               record. Implicitly invokes --record.             
                               Zero or more topics can be specified by          
                               using multiple --record-topic options.           
                               Regular expressions can be used, which           
                               likely requires quotes. A default set of         
                               topics are also recorded, which support          
                               simulation state playback. Enable debug          
                               console output with the -v 4 option              
                               and look for 'Recording default topic' in        
                               order to determine the default set of            
                               topics.                                          
                               Examples:                                        
                                 1. Record all topics.                          
                                     --record-topic ".*"                      
                                 2. Record only the /stats topic.               
                                     --record-topic /stats                      
                                 3. Record the /stats and /clock topics.        
                                     --record-topic /stats                     
[... 53 lines cut ...]

  -z [arg]                     Update rate in Hertz.                            

  -h [--help]                Print this help message.
                                                    
  --force-version <VERSION>  Use a specific library version.
                                                    
  --versions                 Show the available versions.

Environment variables:                                                          
  GZ_SIM_RESOURCE_PATH         Colon separated paths used to locate             
 resources such as worlds and models.                                         

  GZ_SIM_SYSTEM_PLUGIN_PATH    Colon separated paths used to                    
 locate system plugins.                                                       

  GZ_SIM_SERVER_CONFIG_PATH    Path to server configuration file.             

  GZ_GUI_PLUGIN_PATH           Colon separated paths used to locate GUI         
[... cut ...]
  GZ_GUI_RESOURCE_PATH    Colon separated paths used to locate GUI              
 resources such as configuration files.                                       
```

I would maybe also suggest using a different robot description as an example, to use the rrbot model you will need to convert the xacro to URDF, which equires xacro and thus ROS (as far as I can tell). While the tutorial explaining this is already linked in the documentation I think it might be nicer to have a URDF in the accompanying pictures you can just download directly in the URDF format.

audrow, 2022-09-13:
Hey @gaileyleseman, if you can, would you create an issue for this on the corresponding repo or gz-sim?

gaileyleseman, 2022-09-13:
Hi @audrow, I do not think that this is a bug, I assume the functionality is still there, but just moved elsewhere? I've encountered something similar with `gz topic` not being available. Should I still create an issue or maybe post to https://answers.gazebosim.org/questions/?

audrow, 2022-09-14:
@gaileyleseman, Yes, I think it would be good to make an issue for this, since it's not working as intended.

caguero, 2022-09-23:
It should be fixed in the next prerelease after this patch.

nkoenig, 2022-09-23:
@gaileyleseman. Thank you for helping with the tutorial party. Could you please contact me at nate@openrobotics.org so that we can send you a gift?