gazebosim/garden-tutorial-party #2083: gz-math: Vector example

audrow, 2022-09-06:
## Setup
- Os: Ubuntu Jammy
- BuildType: Source

## Links
- [Live](https://gazebosim.org/api/math/7/vector.html)
- [Source](https://github.com/gazebosim/gz-math/tree/gz-math7/tutorials/vector.md)

## Checks
- [ ] **Instructions work**
- [ ] **Images (if there are any) match the result**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

cristinaluna, 2022-09-16:
I'm on it

cristinaluna, 2022-09-16:
Everything works as expected! ;)

audrow, 2022-09-16:
Thank you, @cristinaluna!

closed 2022-09-16 by audrow

nkoenig, 2022-09-23:
@cristinaluna . Thank you for helping with the tutorial party. Could you please contact me at nate@openrobotics.org so that we can send you a gift?